etiennedub/pyk4a #155: How can I use transformed_depth or transformed_depth_point_cloud without interpolation

Qiu731, 2021-12-13:
I find that when I use transformed_depth or transformed_depth_point_cloud , the depth map or point cloud will be interpolated. I want to know how to get the transformed depth map or transformed point cloud without interpolation.

shagren, 2021-12-13:
https://github.com/etiennedub/pyk4a/blob/905d77bfecc0ec425a6335dcddbd63cb09314334/pyk4a/transformation.py#L19

Try to use this with `interp_nearest = False`

shagren, 2021-12-13:
Oh now, it will works only form IR images

lpasselin, 2021-12-14:
I think the only place where we can use interpolation type in the SDK is only a choice between nearest and linear interpolation: `k4a_transformation_depth_image_to_color_camera_custom`. Unfortunately we don't have a "no_interpolation" available with the kinect azure sdk.

https://github.com/etiennedub/pyk4a/blob/8a2fb92f73067b97c35d82b337bc0d0920ffb5b7/pyk4a/pyk4a.cpp#L691

You would need to ask the sdk devs to add this option. Or write your own point cloud creation function based on field of view to get x and y axis?

closed 2021-12-14 by lpasselin