etiennedub/pyk4a #157: Bundle Azure Kinect SDK with PyPi wheel

cansik, 2021-12-22:
As I experienced it myself, there are a lot of problems with finding the dependencies, even if the path to the DLL folder seems correct on Windows.

Is there a reason not to bundle the Azure Kinect SDK (about 2.66 MB) within the wheel?
I guess for deploying applications and easy-of-use this would help a lot.

lpasselin, 2021-12-22:
I agree it is still a problem.

We would have to bundle both Linux and Windows binaries. Or have a package for each platform.

Mo

cansik, 2021-12-25:
I have spent some hours to create a [github actions](https://github.com/cansik/pyk4a/blob/master/.github/workflows/build.yml) and [some adaptions](https://github.com/cansik/pyk4a) to bundle the binary files with the pyk4a releases.

You can test it out by installing the library from the releases:
```
pip install pyk4a --find-links https://github.com/cansik/pyk4a/releases/tag/1.3.0-bundle
```

Currently I have built linux arm64 and windows arm64 and upload the wheels into a release: https://github.com/cansik/pyk4a/releases

It is not yet perfect because of the library naming under unix (contains the version number), but I guess we could fix that with a more intelligent regex. At least the [devices.py](https://github.com/etiennedub/pyk4a/blob/master/example/devices.py) has been tested on Windows & Linux without the Azure Kinect framework installed.

If it makes sense for you, I could open a PR where we can adjust the build process to fit within your strategy.

lpasselin, 2021-12-26:
Awesome!
I am afk for a few days but will make this a priority when I am back.

lpasselin, 2022-07-18:
Let's finish the discussion in the associated PR.
https://github.com/etiennedub/pyk4a/pull/160