etiennedub/pyk4a #159: segmentation fault while using _convert_3d_to_3d, _convert_2d_to_3d

crypto-g245, 2021-12-27:
Hello, 

I wanted to transform the pixel from colour (cx,cy) to depth (dx,dy) cam's view. 
I was using these to function as an intermediate step, but it crashes while using these functions. 

is there any approach to get the desired results (transform the pixel location from colour camera to depth camera's view) or any fixes for the mentioned problem?  

Thanks in Advance.

lpasselin, 2021-12-27:
Hi, could you share a short script to reproduce the problem?

crypto-g245, 2021-12-28:
Hello @lpasselin ,

  Here's the snippet to reproduce the probelm. 

`k4aMaster = PyK4A(config=configs)`
`depth = DepthMode(2)`
`res = ColorResolution(2)`

`calib = Calibration(k4aMaster.device_handle, depth_mode=depth, color_resolution=res) #,thread_safe=False)`

`source_point_3d = (256,846)`
`d = 3586 #mm`
`source_camera = CalibrationType.COLOR`
`target_camera = CalibrationType.DEPTH`
`out = calib.convert_2d_to_3d(source_point_3d,d, source_camera, target_camera)`

rajkundu, 2022-03-12:
I have had segmentation faults occur due to the color image being in compressed/MJPEG form, whereas the underlying C SDK requires a BGRA32 image/array. Perhaps that is what is happening here. Also, what does your `configs` variable look like?

lpasselin, 2022-07-18:
Reopen if you still have this problem.