etiennedub/pyk4a #162: missing system timestamp for depth and color image during playback

maxtob, 2022-01-17:
Hello,

I am having issues with accessing the system timestamps for color and depth images of a MKV file recorded with the pyk4a library.

When reading a MKV recording via
`playback = PyK4APlayback(filename)`
`playback.open()`
`capture = playback.get_next_capture()`

I am able to get the device timestamps for color and depth images
`capture.color_timesamp_usec`
`capture.depth_timesamp_usec`

the system timestamps for both images are, however, returning a 0
`capture.color_system_timestamp_nsec`
`capture.depth_system_timestamp_nsec`

I am able to access both types of timestamps during recording
`device.start()`
`record = PyK4ARecord(device=device, config=config, path=filepath)`
`record.create()`
`capture = device.get_capture()`
`capture.color_timesamp_usec `
`capture.depth_timesamp_usec `
`capture.color_system_timestamp_nsec`
`capture.depth_system_timestamp_nsec`
 `record.write_capture(capture)`

Am I doing something wrong or is this a bug?

Thanks for you help!
Max

shagren, 2022-01-17:
# System Timestamp

https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gaf18827997835a77ac041d5a87203e5ef.html#gaf18827997835a77ac041d5a87203e5ef

> The system timestamp is a high performance and increasing clock (from boot). The timestamp represents the time immediately after the image buffer was read by the host PC.

But this information is not stored/loaded from mkv. Only device timestamp is here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/d87ef578676c05b9a5d23c097502942753bf3777/src/record/internal/matroska_read.cpp#L1865

maxtob, 2022-01-18:
Thanks for the quick reply!

closed 2022-01-18 by lpasselin