etiennedub/pyk4a #163: Unable to set LD_LIBRARY_PATH for pyk4a

yuukireina05, 2022-02-11:
**My Desktop information**

- OS with Version: Ubuntu 20.04.3 LTS 64-bit
- CPU: Intel® Core™ i7-10700 CPU @ 2.90GHz × 16 (No independent GPU)
- SDK Version: 1.4.1
- Packages version: libk4a1.4, libk4abt1.1.0, k4a-tools1.4.1
- `k4aviewer` runs normally on my desktop.

I installed `pyk4a` by calling the command `pip install pyk4a` and required packages by calling `make setup` from the readme of the repo. Also I am sure that `libk4a.so` is in the `bin` of the SDK directory. But it still fails to import module `pyk4a`. 

The log follows:

```
nmed@nmed-OptiPlex-5080:~$ $LD_LIBRARY_PATH/
bash: /home/nmed/Documents/Code/Azure-Kinect-Sensor-SDK/build/bin/: Is a directory
nmed@nmed-OptiPlex-5080:~$ python3
Python 3.8.10 (default, Nov 26 2021, 20:14:08) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyk4a
RuntimeError: module compiled against API version 0xf but this version of numpy is 0xd
Traceback (most recent call last):
  File "/home/nmed/Documents/Code/Azure-Kinect-python/pyk4a/pyk4a/module.py", line 5, in <module>
    import k4a_module  # noqa: F401
ImportError: numpy.core.multiarray failed to import

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/nmed/Documents/Code/Azure-Kinect-python/pyk4a/pyk4a/__init__.py", line 1, in <module>
    from .calibration import Calibration, CalibrationType
  File "/home/nmed/Documents/Code/Azure-Kinect-python/pyk4a/pyk4a/calibration.py", line 8, in <module>
    from .module import k4a_module
  File "/home/nmed/Documents/Code/Azure-Kinect-python/pyk4a/pyk4a/module.py", line 24, in <module>
    raise ImportError(
ImportError: Cannot import k4a_module. Make sure `libk4a.so` can be found. Add the directory to your `LD_LIBRARY_PATH` if required. Also make sure pyk4a is properly built.
```

Build log of the repo follows:

```
nmed@nmed-OptiPlex-5080:~/Documents/Code/Azure-Kinect-python/pyk4a$ make setup
pip install -r requirements-dev.txt
Ignoring dataclasses: markers 'python_version < "3.7"' don't match your environment
Requirement already satisfied: black==19.10b0 in /home/nmed/.local/lib/python3.8/site-packages (from -r requirements-dev.txt (line 1)) (19.10b0)
Requirement already satisfied: flake8==3.8.3 in /home/nmed/.local/lib/python3.8/site-packages (from -r requirements-dev.txt (line 2)) (3.8.3)
Requirement already satisfied: isort==5.4.2 in /home/nmed/.local/lib/python3.8/site-packages (from -r requirements-dev.txt (line 3)) (5.4.2)
Requirement already satisfied: flake8-isort==4.0.0 in /home/nmed/.local/lib/python3.8/site-packages (from -r requirements-dev.txt (line 4)) (4.0.0)
Requirement already satisfied: mypy==0.782 in /home/nmed/.local/lib/python3.8/site-packages (from -r requirements-dev.txt (line 5)) (0.782)
Requirement already satisfied: mypy-extensions==0.4.3 in /home/nmed/.local/lib/python3.8/site-packages (from -r requirements-dev.txt (line 6)) (0.4.3)
Requirement already satisfied: pytest==6.0.1 in /home/nmed/.local/lib/python3.8/site-packages (from -r requirements-dev.txt (line 7)) (6.0.1)
Requirement already satisfied: pytest-cov==2.10.1 in /home/nmed/.local/lib/python3.8/site-packages (from -r requirements-dev.txt (line 8)) (2.10.1)
Requirement already satisfied: typed-ast>=1.4.0 in /home/nmed/.local/lib/python3.8/site-packages (from black==19.10b0->-r requirements-dev.txt (line 1)) (1.5.2)
Requirement already satisfied: toml>=0.9.4 in /home/nmed/.local/lib/python3.8/site-packages (from black==19.10b0->-r requirements-dev.txt (line 1)) (0.10.2)
Requirement already satisfied: regex in /home/nmed/.local/lib/python3.8/site-packages (from black==19.10b0->-r requirements-dev.txt (line 1)) (2022.1.18)
Requirement already satisfied: attrs>=18.1.0 in /usr/lib/python3/dist-packages (from black==19.10b0->-r requirements-dev.txt (line 1)) (19.3.0)
Requirement already satisfied: click>=6.5 in /usr/lib/python3/dist-packages (from black==19.10b0->-r requirements-dev.txt (line 1)) (7.0)
Requirement already satisfied: pathspec<1,>=0.6 in /home/nmed/.local/lib/python3.8/site-packages (from black==19.10b0->-r requirements-dev.txt (line 1)) (0.9.0)
Requirement already satisfied: appdirs in /home/nmed/.local/lib/python3.8/site-packages (from black==19.10b0->-r requirements-dev.txt (line 1)) (1.4.4)
Requirement already satisfied: pyflakes<2.3.0,>=2.2.0 in /home/nmed/.local/lib/python3.8/site-packages (from flake8==3.8.3->-r requirements-dev.txt (line 2)) (2.2.0)
Requirement already satisfied: pycodestyle<2.7.0,>=2.6.0a1 in /home/nmed/.local/lib/python3.8/site-packages (from flake8==3.8.3->-r requirements-dev.txt (line 2)) (2.6.0)
Requirement already satisfied: mccabe<0.7.0,>=0.6.0 in /home/nmed/.local/lib/python3.8/site-packages (from flake8==3.8.3->-r requirements-dev.txt (line 2)) (0.6.1)
Requirement already satisfied: testfixtures<7,>=6.8.0 in /home/nmed/.local/lib/python3.8/site-packages (from flake8-isort==4.0.0->-r requirements-dev.txt (line 4)) (6.18.3)
Requirement already satisfied: typing-extensions>=3.7.4 in /home/nmed/.local/lib/python3.8/site-packages (from mypy==0.782->-r requirements-dev.txt (line 5)) (4.0.1)
Requirement already satisfied: iniconfig in /home/nmed/.local/lib/python3.8/site-packages (from pytest==6.0.1->-r requirements-dev.txt (line 7)) (1.1.1)
Requirement already satisfied: more-itertools>=4.0.0 in /home/nmed/.local/lib/python3.8/site-packages (from pytest==6.0.1->-r requirements-dev.txt (line 7)) (8.12.0)
Requirement already satisfied: pluggy<1.0,>=0.12 in /home/nmed/.local/lib/python3.8/site-packages (from pytest==6.0.1->-r requirements-dev.txt (line 7)) (0.13.1)
Requirement already satisfied: py>=1.8.2 in /home/nmed/.local/lib/python3.8/site-packages (from pytest==6.0.1->-r requirements-dev.txt (line 7)) (1.11.0)
Requirement already satisfied: packaging in /home/nmed/.local/lib/python3.8/site-packages (from pytest==6.0.1->-r requirements-dev.txt (line 7)) (21.3)
Requirement already satisfied: coverage>=4.4 in /home/nmed/.local/lib/python3.8/site-packages (from pytest-cov==2.10.1->-r requirements-dev.txt (line 8)) (6.3.1)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in /usr/lib/python3/dist-packages (from packaging->pytest==6.0.1->-r requirements-dev.txt (line 7)) (2.4.6)
```

I set the `LD_LIBRARY_PATH` by adding `export LD_LIBRARY_PATH=/home/nmed/Documents/Code/Azure-Kinect-Sensor-SDK/build/bin` to `~/.bashrc`. I also tried replacing the path with `/usr/lib/x86_64-linux-gnu` but still got the same import error. 

I would appreciate it if you could kindly give your suggestion on it help me solving the problem. Thank you!

leifvan, 2022-03-13:
Hi @yuukireina05, in case you have not solved your problem by now, I ran into this issue multiple times (albeit using Windows). The important line from your log is
```
RuntimeError: module compiled against API version 0xf but this version of numpy is 0xd
```
which seems to indicate an incompatible version of NumPy. For me it sufficed to
* uninstall pyk4a,
* update NumPy to the current version ``1.22.3``,
* install pyk4a version ``1.3.0`` again.

yuukireina05, 2022-03-21:
@leifvan Thank you for your reply! It finally worked by following your suggestion. 
Now I can import `pyk4a` without any problem.

closed 2022-03-21 by yuukireina05