etiennedub/pyk4a #164: Error with transformations in recorded file (playback): color frame not BGRA32

rajkundu, 2022-03-07:
Hello! I was trying to run a few of the examples, such as `/examples/viewer_point_cloud.py`, but with pre-recorded `.mkv` files instead of in a "live" session with a device attached. However, an exception was being thrown because the `.mkv` file was recorded in MJPG format – whereas the transformation functions needed a BGRA32-formatted color frame.

I ended up fixing this issue myself and am adding the solution here for anyone in a similar situation, because even though it was incredibly simple to solve, it did take me a little while. [In my own fork (4 most recent commits)](https://github.com/rajkundu/pyk4a), I ended up adding a `bool` parameter called `force_bgra`, and I made the `pyk4a.capture.color` property getter convert the color frame to BGRA32 as necessary. Now, I can create a `PyK4APlayback` object, passing `force_bgra=True`, then run the rest of `viewer_point_cloud.py` without error:

```python
# Open recording/playback
playback = PyK4APlayback("./recording.mkv", force_bgra=True)
playback.open()

# Get first frame with depth and color
while True:
	capture = playback.get_next_capture()
	if np.any(capture.depth) and np.any(capture.color):
		break
points = capture.depth_point_cloud.reshape((-1, 3))
colors = capture.transformed_color[..., (2, 1, 0)].reshape((-1, 3))

# Plot point cloud with color
fig = plt.figure()
ax = fig.add_subplot(111, projection="3d")
ax.scatter(
    points[:, 0], points[:, 1], points[:, 2], s=1, c=colors / 255,
)
ax.set_xlabel("x")
ax.set_ylabel("y")
ax.set_zlabel("z")
ax.set_xlim(-2000, 2000)
ax.set_ylim(-2000, 2000)
ax.set_zlim(0, 4000)
ax.view_init(elev=-90, azim=-90)
plt.show()

# Close recording/playback
playback.close()
```

rajkundu, 2022-03-07:
This issue has already been resolved by me, so I'm "closing" it for bookkeeping purposes. However, I have _not_ tested it extensively (only on `viewer_point_cloud.py`), so please feel free to comment with any issues, additions, comments, suggestions, etc!

closed 2022-03-07 by rajkundu

lpasselin, 2022-03-07:
@shagren an you take a look?

shagren, 2022-03-09:
@lpasselin, I think pyk4a do not need this functionality. Decoding images is cpu-cost and not necessary for many cases. But some documentation/example about conversion can be added. I will propose some PR later.

rajkundu, 2022-03-10:
I am not sure what type of research/work is usually done with the Azure Kinect DK devices, but my case, research involving human subjects, requires data to be collected and then analyzed separately, at a later time. So, nearly all of my work is done with recorded data, not live data.

In my opinion, every operation that can be done live (e.g., transformations, reconstruction, etc.) should also be able to be done on a recorded sequence; there should be practically no difference between live vs. recorded data other than the source being device vs. hard disk.

I do think that this would make a good addition to the library because it is not harming anything. I mean, it is off by default, and it simply gives users the *option* of being able to call BGRA-requiring functions without having to add to the source code themselves, like I did. `pyk4a` is a great library, and I'd love to be able to help make it even better for others!

lpasselin, 2022-03-10:
Thanks for looking at this Shagren.
rajkundu thanks for taking the time for sharing and explaining your point.

I agree with rajkundu that we should implement this. Enabling operations is good to have and can be really useful. Also playback of lossless recording is required for many use cases. Personally I would have never recorded in lossy format for my research.

With that being said, I did not look at the code yet and my understanding of the playback code is limited.

We should make a PR. Ideally with some tests but they can come in a future PR since you confirmed it works.

rajkundu, 2022-03-10:
@lpasselin Of course! Glad to help.

One question: I would also love to record raw color data, but based on the following table:

RGB Camera Resolution (HxV) | Aspect Ratio | Format Options | Frame Rates (FPS) | Nominal FOV (HxV)(post-processed)
-- | -- | -- | -- | --
3840x2160 | 16:9 | MJPEG | 0, 5, 15, 30 | 90°x59°
2560x1440 | 16:9 | MJPEG | 0, 5, 15, 30 | 90°x59°
1920x1080 | 16:9 | MJPEG | 0, 5, 15, 30 | 90°x59°
1280x720 | 16:9 | MJPEG/YUY2/NV12 | 0, 5, 15, 30 | 90°x59°
4096x3072 | 4:3 | MJPEG | 0, 5, 15 | 90°x74.3°
2048x1536 | 4:3 | MJPEG | 0, 5, 15, 30 | 90°x74.3°

taken from [this page](https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification#color-camera-supported-operating-modes), it seems to me that the Azure Kinect DK isn't really capable of doing so – it appears that the device compresses the raw color frame as MJPG before sending it to the host computer. Then, this compressed MJPG color frame can be decompressed (i.e., with loss) into BGRA32 if needed. So, when you read a BGRA32 image from the SDK, it is my understanding that it is not actually raw, lossless sensor data, but rather data that has been compressed into MJPG and then decompressed, with loss, back into a BGRA32 array of fixed size.

What do you think? What is your interpretation of this table and the rest of the device specifications?

rajkundu, 2022-06-15:
As a sidenote, I noticed recently that whenever I try to record in BGRA32 format at 2160p, my computer cannot keep up; I get the "disk write speed too low" error message, and my memory/RAM instantly begins to fill up.

Is this normal behavior - have you all experienced this as well? Do you record in BGRA32 or MJPG, and at what resolution? It is a decently powerful laptop (11th-gen i7, 3080 Ti), which is why I am a bit surprised.

rajkundu, 2022-06-15:
No worries - I mean recording with `pyk4a`. What happens when you try to run [example/record.py](https://github.com/etiennedub/pyk4a/blob/master/example/record.py), but changing the config on line 12 from `config = Config(color_format=ImageFormat.COLOR_MJPG)` to `config = Config(color_format=ImageFormat.COLOR_BGRA32)`?

lpasselin, 2022-06-15:
I don't have my device and am on a mac unfortunately.

Can you confirm that recording with these settings works with the k4a viewer? I just want to make sure this isn't caused by the k4a SDK.

rajkundu, 2022-06-15:
Great idea, and I should certainly have done that. I'll do it now!

Also, I couldn't find a way to record from inside the k4a viewer; I've been using the [`k4arecorder` command line utility](https://docs.microsoft.com/en-us/azure/kinect-dk/azure-kinect-recorder) to write captures to disk. Do you know if there is a way to record directly from within the viewer? That would be a lot easier, if so!

lpasselin, 2022-06-15:
No idea! I thought it was possible from the viewer but I might be wrong.

rajkundu, 2022-06-15:
Ah, man. It seems there's no way to set the color format with the `k4arecorder` utility. I assume it just captures in MJPG since that's natively transmitted by the camera, and since it's not rendering anything, there's no need to convert to BGRA32. Perhaps someone else can try out `example/record.py` but with `color_format=ImageFormat.COLOR_BGRA32`?

rajkundu, 2022-06-15:
Yes, I guess the issue is that decoding the MJPG into BGRA32 takes some time and seems to be blocking/filling up the write queue. Perhaps if I have time, I will try making a simple application that directly uses the C SDK to record with BGRA32 configuration and then see if it has the same memory issues.

rajkundu, 2022-06-15:
I suppose I could also just [slightly modify the source code](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/tools/k4arecorder/main.cpp#L116) and recompile `k4arecorder` to use BGRA32 instead of MJPG in its configuration.

rajkundu, 2022-08-30:
I ended up rewriting my application in C++, and everything works perfectly fine; I can record BGRA32 captures at 30 FPS with no memory leaks – I'm still not sure why I had a leak when using pyk4a.

rajkundu, 2022-09-01:
> Thanks for looking at this Shagren. rajkundu thanks for taking the time for sharing and explaining your point.
>
> I agree with rajkundu that we should implement this. Enabling operations is good to [...]

I realized after looking back at my code that it perhaps is not the best solution to this problem. While I found it convenient to have a flag such as `force_bgra=True`, behind the scenes, this would require `pyk4a` to depend on `cv2`/OpenCV just for this one feature – which is a downright terrible idea.

Instead, if anyone has this same issue (your recording is in MJPG format, but `pyk4a` functions require BGRA32 format), I would suggest manually overwriting the `PyK4ACapture` object's properties as follows:
```python
playback = pyk4a.PyK4APlayback('1662056463_Left.mkv')
playback.open()
while True:
	# Get next capture
	try:
		capture = playback.get_next_capture()
	except EOFError:
		break

	# Convert MJPG color image to BGRA32
	capture._color = cv2.cvtColor(cv2.imdecode(capture.color, cv2.IMREAD_COLOR), cv2.COLOR_BGR2BGRA)
	capture._color_format = pyk4a.ImageFormat.COLOR_BGRA32
	
	# Now we can use the capture object as if it were originally recorded in BGRA32 format!
	capture.transformed_color # e.g. this function works without error

playback.close()
```
Perhaps it's worth officially documenting this workaround somewhere, or perhaps not – it seems like nobody else has really encountered these issues. Hope this helps anyone who needs it, though!