etiennedub/pyk4a #166: Misalignment of depth_image_to_color_camera?

kdquine, 2022-05-04:
Hi, I've been trying to use pyk4a.depth_image_to_color_camera, but the end result winds up being both displaced and distorted relative to the color camera image, while being the correct resolution - any advice for figuring out what I'm doing wrong?

I get the raw calibration JSON from the mkv and pass that in to pyk4a.Calibration.from_raw() using the modes listed in the embedded xml, and then call pyk4a.depth_image_to_color_camera.  Do we expect the depth and color images to approximately superimpose?

lpasselin, 2022-05-04:
From what I remember, they did approximately match. Can you share examples?

kdquine, 2022-05-05:
Sure, I've uploaded a small clip, some simplified code and some example outputs here:

https://www.dropbox.com/sh/4eewzr5rchczhvw/AAC5ujuv34YS71T5Mmw0bx7Aa?dl=0

Thanks for your help!

meteorshowers, 2022-05-08:
> Hi, I've been trying to use pyk4a.depth_image_to_color_camera, but the end result winds up being both displaced and distorted relative to the color camera image, while being the correct resolution - [...]

the same with you

lpasselin, 2022-05-09:
To me that does seem a bit too much offset from depth to color. With that being said, I never tried capturing with NFOV...

Can you try to display live captures? I worry the error might be related to the playback from captured video.

etiennedub, 2022-05-10:
>  I get the raw calibration JSON from the mkv and pass that in to pyk4a.Calibration.from_raw().

Can you try to get the calibration from [save_calibration_json](https://github.com/etiennedub/pyk4a/blob/master/pyk4a/pyk4a.py#L73)? Other than that, do you have the same issue with an other mode instead of NFOV?

If it's really a factory calibration issue, you might have to manually calibrate it.

kdquine, 2022-05-10:
> > I get the raw calibration JSON from the mkv and pass that in to pyk4a.Calibration.from_raw().
> 
> Can you try to get the calibration from [save_calibration_json](https://github.com/etiennedub/pyk4a/blob/master/pyk4a/pyk4a.py#L73)?

The json comes from the mkv file (extracted with mkvextract) and is put there by k4arecorder. I don't think it's an issue with the json; the only reason I packaged it as a separate file is that I didn't want to have to include mkvextract with the sample.

> Other than that, do you have the same issue with an other mode instead of NFOV?

I'll give WFOV a try.

> If it's really a factory calibration issue, you might have to manually calibrate it.

Is that a thing? I don't recall seeing anything about manual calibration of the depth + color camera?

etiennedub, 2022-05-10:
For the JSON, it was just an idea to isolate the problem. I would still try to export the JSON and compare if it's the same from the MKV. `k4arecorder` probably save the same think but I just don't know how it works.

I think they calibrate it in factory and store it in the camera (basically the JSON you can export). Otherwise they just store generic parameters for all cameras. In both case, you can have some gain with a custom calibration. But in your case, the transformation is so off, that I doubt it's the issue. 

If that can help, example of overlay depth that I got in the past (sorry but no clue on parameters and mode):
[image]

closed 2022-07-18 by lpasselin