etiennedub/pyk4a #172: pip install fails on Ubuntu 20.04.4

rajkundu, 2022-06-11:
Hello! I'm trying to figure out why `pip install pyk4a` doesn't work on my Linux machine. Whenever I try to run it, it fails and prints the following output:

```
(tms_base) raj@taco:~$ pip install pyk4a
Collecting pyk4a
  Using cached pyk4a-1.4.0.tar.gz (23 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: numpy in ./miniconda3/envs/tms_base/lib/python3.9/site-packages (from pyk4a) (1.22.4)
Requirement already satisfied: python-version>="3.4" in ./miniconda3/envs/tms_base/lib/python3.9/site-packages (from pyk4a) (0.0.2)
Building wheels for collected packages: pyk4a
  Building wheel for pyk4a (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/raj/miniconda3/envs/tms_base/bin/python /home/raj/miniconda3/envs/tms_base/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmpdpkftl8_
       cwd: /tmp/pip-install-cadgcu2m/pyk4a_81b24334acf54a2a86b31781d2ac6b27
  Complete output (39 lines):
  /tmp/pip-build-env-zymqyr5f/overlay/lib/python3.9/site-packages/setuptools/dist.py:772: UserWarning: Usage of dash-separated 'description-file' will not be supported in future versions. Please use the underscore name 'description_file' instead
    warnings.warn(
  Automatic kinect SDK detection did not yield any results.
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-cpython-39
  creating build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/calibration.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/errors.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/config.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/win32_utils.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/transformation.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/playback.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/results.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/module.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/pyk4a.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/__init__.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/capture.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/record.py -> build/lib.linux-x86_64-cpython-39/pyk4a
  copying pyk4a/py.typed -> build/lib.linux-x86_64-cpython-39/pyk4a
  running build_ext
  building 'k4a_module' extension
  creating build/temp.linux-x86_64-cpython-39
  creating build/temp.linux-x86_64-cpython-39/pyk4a
  gcc -pthread -B /home/raj/miniconda3/envs/tms_base/compiler_compat -Wno-unused-result -Wsign-compare -DNDEBUG -O2 -Wall -fPIC -O2 -isystem /home/raj/miniconda3/envs/tms_base/include -I/home/raj/miniconda3/envs/tms_base/include -fPIC -O2 -isystem /home/raj/miniconda3/envs/tms_base/include -fPIC -I/tmp/pip-build-env-zymqyr5f/overlay/lib/python3.9/site-packages/numpy/core/include -I/home/raj/miniconda3/envs/tms_base/include/python3.9 -c pyk4a/pyk4a.cpp -o build/temp.linux-x86_64-cpython-39/pyk4a/pyk4a.o
  In file included from /tmp/pip-build-env-zymqyr5f/overlay/lib/python3.9/site-packages/numpy/core/include/numpy/ndarraytypes.h:1960,
                   from /tmp/pip-build-env-zymqyr5f/overlay/lib/python3.9/site-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                   from /tmp/pip-build-env-zymqyr5f/overlay/lib/python3.9/site-packages/numpy/core/include/numpy/arrayobject.h:5,
                   from pyk4a/pyk4a.cpp:2:
  /tmp/pip-build-env-zymqyr5f/overlay/lib/python3.9/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
     17 | #warning "Using deprecated NumPy API, disable it with " \
        |  ^~~~~~~
  pyk4a/pyk4a.cpp:4:10: fatal error: k4a/k4a.h: No such file or directory
      4 | #include <k4a/k4a.h>
        |          ^~~~~~~~~~~
  compilation terminated.
  error: command '/usr/bin/gcc' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for pyk4a
Failed to build pyk4a
ERROR: Could not build wheels for pyk4a which use PEP 517 and cannot be installed directly
(tms_base) raj@taco:~$ echo $LD_LIBRARY_PATH
/usr/lib/x86_64-linux-gnu/
(tms_base) raj@taco:~$ ls -la /usr/lib/x86_64-linux-gnu/ | grep k4a
drwxr-xr-x   2 root root      4096 Jun 10 17:08 libk4a1.4
lrwxrwxrwx   1 root root        19 Jun 15  2020 libk4arecord.so -> libk4arecord.so.1.4
lrwxrwxrwx   1 root root        21 Jun 15  2020 libk4arecord.so.1.4 -> libk4arecord.so.1.4.1
-rw-r--r--   1 root root  13810400 Jun 15  2020 libk4arecord.so.1.4.1
lrwxrwxrwx   1 root root        13 Jun 15  2020 libk4a.so -> libk4a.so.1.4
lrwxrwxrwx   1 root root        15 Jun 15  2020 libk4a.so.1.4 -> libk4a.so.1.4.1
-rw-r--r--   1 root root   6916616 Jun 15  2020 libk4a.so.1.4.1
(tms_base) raj@taco:~$ 
```
How might I go about fixing this? Thank you for your help!

rajkundu, 2022-06-11:
🤦 It was because I didn't install `libk4a1.4-dev`:
```
(tms_base) raj@taco:~$ sudo apt install libk4a1.4-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfwupdplugin1
Use 'sudo apt autoremove' to remove it.
The following NEW packages will be installed:
  libk4a1.4-dev
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 44.1 kB of archives.
After this operation, 313 kB of additional disk space will be used.
Get:1 https://packages.microsoft.com/ubuntu/18.04/prod bionic/main amd64 libk4a1.4-dev amd64 1.4.1 [44.1 kB]
Fetched 44.1 kB in 0s (224 kB/s)         
Selecting previously unselected package libk4a1.4-dev.
(Reading database ... 167230 files and directories currently installed.)
Preparing to unpack .../libk4a1.4-dev_1.4.1_amd64.deb ...
Unpacking libk4a1.4-dev (1.4.1) ...
Setting up libk4a1.4-dev (1.4.1) ...
(tms_base) raj@taco:~$ pip install pyk4a
Collecting pyk4a
  Using cached pyk4a-1.4.0.tar.gz (23 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting python-version>="3.4"
  Using cached python_version-0.0.2-py2.py3-none-any.whl (3.4 kB)
Requirement already satisfied: numpy in ./miniconda3/envs/tms_base/lib/python3.9/site-packages (from pyk4a) (1.22.4)
Building wheels for collected packages: pyk4a
  Building wheel for pyk4a (PEP 517) ... done
  Created wheel for pyk4a: filename=pyk4a-1.4.0-cp39-cp39-linux_x86_64.whl size=34013 sha256=e5c59a3ac958cc9f39046a5117a1731b03236cb5b5bb024a5ba3b9123b294f54
  Stored in directory: /home/raj/.cache/pip/wheels/db/ac/3b/fca70419aed86529018fe7968808b74b54d0ae7b8d5b3efb8e
Successfully built pyk4a
Installing collected packages: python-version, pyk4a
Successfully installed pyk4a-1.4.0 python-version-0.0.2
(tms_base) raj@taco:~$ 
```
Once I did, installation worked flawlessly! Thank you!

I think that when I originally installed the SDK, I thought the `dev` packages were development/beta versions or something, so I omitted them. I didn't realize until looking closer at the [SDK installation instructions](https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download#linux-installation-instructions) that "the `libk4a<major>.<minor>-dev` package contains the headers and CMake files to build your applications/executables against `libk4a`." Sorry for filing an issue over nothing!

lpasselin, 2022-06-11:
That's something we should add in the readme for sure. I really didn't know what the problem was.

Let's leave open until we add this info to the readme.

rajkundu, 2022-06-15:
Reopening as per request above

lpasselin, 2022-08-04:
Added info in readme with this PR https://github.com/etiennedub/pyk4a/pull/176