etiennedub/pyk4a #173: Error installing the py4ak lib

AbdelRahmanNasser20, 2022-07-10:
When I try to pip install the py4ak lib using the following command:

`
pip install pyk4a --no-use-pep517 --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"
`

I get the following error:

```
Running setup.py install for pyk4a ... error
  error: subprocess-exited-with-error

  × Running setup.py install for pyk4a did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Automatically detected kinect SDK. Adding include dir: C:\Program Files\Azure Kinect SDK v1.4.1\sdk\include and library dir C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\AMD64\release\lib.
      c:\users\abdel nasser\desktop\python\python38\lib\site-packages\setuptools\dist.py:771: UserWarning: Usage of dash-separated 'description-file' will not be supported in future versions. Please use the underscore name 'description_file' instead
        warnings.warn(
      running build_ext
      building 'k4a_module' extension
      error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pyk4a

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

I have the following path added

`
I have added the following path to my environment variables.
C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin
`

[Image of C++ build tools that are installed][1]

  [1]: https://i.stack.imgur.com/7lIHb.jpg

I am not at all sure how to deal with this: 

`
c:\users\abdel nasser\desktop\python\python38\include\pyconfig.h(59): fatal error C1083: Cannot open include file: 'io.h': No such file or directory
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.32.31326\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]
`
Any help will be greatly appreciated.

lpasselin, 2022-07-11:
Not finding io.h is a first for me. I am not an expert with windows. Let's hope an other user can help you with this.

In the meantime, you could try installing using the bundle. pip install pyk4a-bundle

AbdelRahmanNasser20, 2022-07-13:
yes I am also on win11 and just like you `pip install pyk4a-bundle` worked flawlessly.

lpasselin, 2022-07-13:
pyk4a-bundle is precompiled

closed 2022-07-13 by lpasselin