etiennedub/pyk4a #175: Get Exposure and ISO values in playback

hobbitsyfeet, 2022-07-26:
I am looking to grab the Exposure and ISO for each frame through the Pyk4aPlayback. In C++ it would look something like this:

```
capture = k4a_playback_get_next_capture()
image = k4a_capture_get_colour_image() # Returns Image object k4a_image_t
exposure = k4a_image_get_exposure_usec(image)
iso = k4a_image_get_iso_speed(image)
```

With PyK4a, you access the image through `playback.get_next_capture().color` which ends up being a NumPy array of the colour image with no access to exposure or ISO values.

I saw the function in pyk4a.cpp and wondered if you are moving towards implementation.
```
static PyObject *color_image_get_exposure_usec(PyObject *self, PyObject *args) {
  k4a_capture_t *capture_handle;
  PyObject *capsule;
  uint64_t exposure_usec = 0;
  PyArg_ParseTuple(args, "O", &capsule);
  capture_handle = (k4a_capture_t *)PyCapsule_GetPointer(capsule, CAPSULE_CAPTURE_NAME);

  k4a_image_t image = k4a_capture_get_color_image(*capture_handle);
  if (image == NULL) {
    fprintf(stderr, "Color image missed");
    return Py_BuildValue("K", exposure_usec);
  }

  exposure_usec = k4a_image_get_exposure_usec(image);
  k4a_image_release(image);
  return Py_BuildValue("K", exposure_usec);
}
```

Thanks.

lpasselin, 2022-07-27:
Hi,

With pyk4a, all capture data is available with the PyK4ACapture object. 
````python
capture: PyK4ACapture = playback.get_next_capture()
print(capture.color_exposure_usec)
````
See here: https://github.com/etiennedub/pyk4a/blob/2791f6ea6678f889e3d6a730eb924fd7cd4a95ee/pyk4a/capture.py#L68

iso was not implemented and I am not sure why. It should be easy to add. Almost the same code as color_usec retrieval. You should create a PR if you implement it!

hobbitsyfeet, 2022-07-27:
Brilliant! I found I was using a version from 2020 and this was implemented since then.

I'll look into mimicking your code to get iso as well (I'm not experienced, is there somewhere you could point to get familiar with porting the code?)

Thank you very much!

hobbitsyfeet, 2022-07-28:
Ok, I got it working after writing and compiling it. Do you have any knowledge if MKV files record exposure and ISO? The functions seem to return 0, and raise the "Cannot read exposure from color image" so my guess is not.

I'll write up a PR in a bit with tests.

lpasselin, 2022-07-28:
> Do you have any knowledge if MKV files record exposure and ISO? The functions seem to return 0, and raise the "Cannot read exposure from color image" so my guess is not.

I really don't know. Maybe @shagren  can answer

closed 2022-07-28 by lpasselin

shagren, 2022-08-01:
i dunno :(

lpasselin, 2022-08-04:
Associateed PR: https://github.com/etiennedub/pyk4a/pull/176