etiennedub/pyk4a #180: TypeError using with statement to read PyK4APlayback

chrismeunier, 2022-10-28:
Hi I just found a bug when reading recordings using a `with` context. With a simple sample like so:

```python
from pyk4a import PyK4APlayback

with PyK4APlayback("recording.mkv") as p: # call __enter__
    while True:
        try:
            p.get_next_capture()
            # do stuff
        except EOFError:
            break # get out of the with statement -> call __exit__
```

Upon exiting the with statement I get `TypeError: __exit__() takes 1 positional argument but 4 were given`. Which is actually normal because [it takes 4](https://docs.python.org/3/reference/datamodel.html?highlight=__exit__#object.__exit__) even if the others may be `None`.
So I guess it's a simple fix, [`__exit__(self)`](https://github.com/etiennedub/pyk4a/blob/2791f6ea6678f889e3d6a730eb924fd7cd4a95ee/pyk4a/playback.py#L60) just needs to be replaced by `__exit__(self, exc_type, exc_value, traceback)`. I did it locally and it works just fine!

chrismeunier, 2022-10-28:
Oh and while looking for help in playback.py I found a small typo: [`get_previous_capture()`](https://github.com/etiennedub/pyk4a/blob/2791f6ea6678f889e3d6a730eb924fd7cd4a95ee/pyk4a/playback.py#L171) has an extra "e". 😉

lpasselin, 2022-10-28:
Thanks for reporting with a fix! Should be an easy PR.

chrismeunier, 2022-11-01:
Here you go! I'll leave it to you :)

closed 2022-11-01 by chrismeunier