etiennedub/pyk4a #186: How can I get the intrinsic parameters of depth sensor ?

zhoujun-7, 2022-12-17:
I tried to get the intrinsic parameters of depth sensor, which are focal length and principle point. And I run the following code.

```
import pyk4a
from pyk4a import Config, PyK4A
from pprint import pprint

device = PyK4A(
            Config(
                color_resolution=pyk4a.ColorResolution.RES_1080P,
                depth_mode=pyk4a.DepthMode.NFOV_UNBINNED,
                synchronized_images_only=True,
            )
        )
device.start()
device.whitebalance = 4500

data = eval(device.calibration_raw)
pprint(data["CalibrationInformation"]['Cameras'][0])
```

the output is 

```
{'Intrinsics': {'ModelParameterCount': 14,
                'ModelParameters': [0.5048612952232361,
                                    0.4995821714401245,
                                    0.49207374453544617,
                                    0.492192804813385,
                                    2.1669528484344482,
                                    1.4007930755615234,
                                    0.07497162371873856,
                                    2.5014450550079346,
                                    2.0837771892547607,
                                    0.3902532160282135,
                                    0,
                                    0,
                                    2.1939287762506865e-05,
                                    -2.953955117845908e-06],
                'ModelType': 'CALIBRATION_LensDistortionModelBrownConrady'},
 'Location': 'CALIBRATION_CameraLocationD0',
 'MetricRadius': 1.739999771118164,
 'Purpose': 'CALIBRATION_CameraPurposeDepth',
 'Rt': {'Rotation': [1, 0, 0, 0, 1, 0, 0, 0, 1], 'Translation': [0, 0, 0]},
 'SensorHeight': 1024,
 'SensorWidth': 1024,
 'Shutter': 'CALIBRATION_ShutterTypeUndefined',
 'ThermalAdjustmentParams': {'Params': [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]}}
```

But it dosen't seems like the focal length and principle point.

etiennedub, 2023-01-03:
Hi, I think you can call [Calibration.get_camera_matrix](https://github.com/etiennedub/pyk4a/blob/2791f6ea6678f889e3d6a730eb924fd7cd4a95ee/pyk4a/calibration.py#L175) to get an opencv like matrix.

Try `device._calibration.get_camera_matrix(CalibrationType.DEPTH)`.

closed 2023-01-03 by lpasselin

zhoujun-7, 2023-01-08:
It is what I want. 
Thanks for your help.

zhoujun-7, 2023-01-08:
> Hi, I think you can call [Calibration.get_camera_matrix](https://github.com/etiennedub/pyk4a/blob/2791f6ea6678f889e3d6a730eb924fd7cd4a95ee/pyk4a/calibration.py#L175) to get an opencv like matrix.
> 
> Try `device._calibration.get_camera_matrix(CalibrationType.DEPTH)`.

It is what I want.
Thanks for your help.