etiennedub/pyk4a #190: convert_2d_to_3d produces Value Errors

nliedmey, 2023-01-27:
Hello,

I'm using the _convert_2d_to_3d_ function to bring my 2D-Coordinates and thebelonging depth to the 3D-Coordinate system of the camera. I am taking the depth value from using looking up the coordinate's depth value in the _capture.transformed_depth_ image.

Now when throwing the coordinates and their depth values into _convert_2d_to_3d_ function, i get many valid 3D coordinates but also "some" Value Errors. For 208 Frames, with each having 33 Landmark-coordinates, i get 6864 valid and 352 non-valid 3D-Coordinates.

Is this a normal value or is there a way to potimize this? What is the best color resolution & (WFOV) depth mode combination for this purpose or is this not important here?

I looked up the SDK documentation below but this was not very helpful for me:
[https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga664602bdb48dab38117a6c1d14b880de.html#ga664602bdb48dab38117a6c1d14b880de](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga664602bdb48dab38117a6c1d14b880de.html#ga664602bdb48dab38117a6c1d14b880de)

Thank you for your helping me out :-)
Greets

nliedmey, 2023-01-30:
Ok i have found out that missing depth values, which return a value of 0mm, are the reason for ~80% of the non-valid 3D-coordinates. I'll try to minimize them though interpolation.
But what could be the reason for the other non-valid 3D-coordinates?

JJLimmm, 2023-02-08:
Hi @nliedmey , i am having some trouble in using the API convert_2d_to_3d(). Could you show me how did you write the code and input arguments for using this API?

JJLimmm, 2023-02-09:
Hi @lpasselin , Invalid values are caused by the SDK as mentioned in the [online documentation](https://learn.microsoft.com/en-us/azure/kinect-dk/depth-camera#invalidation:~:text=dt.-,Invalidation,-In%20certain%20situations).
Are you able to show a snippet of your code that calls this API? convert2d_to_3d()

nliedmey, 2023-06-28:
Sorry for late response.

I don't know why but i got many depth measurements with value of 120mm that seem to produce 3D calculation error through this SDK..
120mm is outside the operating range of WFOV_Unbinned mode and no realistic values for my use case anyway.

Is this a known issue or has someone else the same problem?

lpasselin, 2023-06-28:
Sorry I have not seen this before. 

The only time I have seen weird values is when my device was sitting on a table, not a tripod. It was sending back weird values due to the obstruction of the table which was in the field of view of the camera!