etiennedub/pyk4a #203: --external-sync flag in pyk4a

timsainb, 2023-02-28:
The [sdk website](https://learn.microsoft.com/en-us/azure/Kinect-dk/record-external-synchronized-units) explains how to use an external trigger (e.g. an arduino) to acquire frames using the --external sync flag.

e.g. 

```
k4arecorder.exe --external-sync sub -e -8 -r 5 -l 10 sub1.mkv
```

How would you do the same in pyk4a?

shagren, 2023-03-01:
Subordinates: 
```
 config1 = Config(..., wired_sync_mode=WiredSyncMode.SUBORDINATE, subordinate_delay_off_master_usec=0)
 config2 = Config(..., wired_sync_mode=WiredSyncMode.SUBORDINATE, subordinate_delay_off_master_usec=200)
 config3 = Config(..., wired_sync_mode=WiredSyncMode.SUBORDINATE, subordinate_delay_off_master_usec=400)
 ....
 device1=PyK4A(config=config, device_id=device_id)
 device1.start()
...
while True:
   capture1 = device1.get_capture()
   capture2 = device2.get_capture()
   ...

```
Each subordinate must have own subordinate_delay_off_master_usec. I do not remember exact value, but each next must be greater by some minimal  value. 200 is suitable in any case, but you can read SDK docs for better value. 

All devices must have same Color and Depth modes, exposure and powerline_frequency settings.

Sometime captures will  be returned from different ticks (external trigger ticks). You can compare `capture.depth_timestamp_usec` with each other, look at https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/examples/green_screen/MultiDeviceCapturer.h .

timsainb, 2023-03-02:
Awesome, thanks!

closed 2023-03-02 by timsainb