etiennedub/pyk4a #208: WFOV_UNBINNED does not work

vstadnytskyi-FDA, 2023-05-01:
Hello,

I am exploring different setting for the depth camera.

To set up different depth parameters

```
    from pyk4a import Config, PyK4A
    # Load camera with the default config
    k4a = PyK4A(   Config(
            depth_mode=pyk4a.config.DepthMode.WFOV_UNBINNED
        )
    )
```
with different option being

- WFOV_UNBINNED,
- WFOV_2x2BINNED, 
- NFOV_UNBINNED,
- NFOV_2x2BINNED

I tried all four and three are working.

When I try 'WFOV_UNBINNED' i get the following error
```
---------------------------------------------------------------------------
K4AException                              Traceback (most recent call last)
Input In [2], in <cell line: 29>()
     27     k4a.stop()
     28 k4a = init()
---> 29 start(k4a)

Input In [2], in start(k4a)
     23 def start(k4a):
---> 24     k4a.start()

File ~\AppData\Roaming\Python\Python38\site-packages\pyk4a\pyk4a.py:36, in PyK4A.start(self)
     34 if not self.opened:
     35     self.open()
---> 36 self._start_cameras()
     37 self._start_imu()
     38 self.is_running = True

File ~\AppData\Roaming\Python\Python38\site-packages\pyk4a\pyk4a.py:94, in PyK4A._start_cameras(self)
     92 def _start_cameras(self):
     93     res = k4a_module.device_start_cameras(self._device_handle, self.thread_safe, *self._config.unpack())
---> 94     _verify_error(res)

File ~\AppData\Roaming\Python\Python38\site-packages\pyk4a\errors.py:18, in _verify_error(res)
     16 res = Result(res)
     17 if res == Result.Failed:
---> 18     raise K4AException()
     19 elif res == Result.Timeout:
     20     raise K4ATimeoutException()

K4AException: 
```

vstadnytskyi-FDA, 2023-05-01:
For example, if I modify example/viewer_depth.py to get WFOV unbinned data I get the same error

lpasselin, 2023-05-01:
Are there any other logs in the console? Sometimes the SDK prints useful error related logs.

vstadnytskyi-FDA, 2023-05-02:
The problem was that I used FPS=30, the default setting, which is not supported by the camera at WFOV_UNBINNED.

closed 2023-05-02 by vstadnytskyi-FDA

lpasselin, 2023-05-02:
Thansk for the explanation. We should add a constraint in the config file to limit these errors.