etiennedub/pyk4a #213: Question about pyk4a.PyK4APlayback ()

xin4431, 2023-07-21:
`def main():

    k4a_playback = pyk4a.PyK4APlayback("C:/AppData/output.mkv")

    k4a_playback.open()
    cv2.namedWindow('Azure Kinect RGB', cv2.WINDOW_NORMAL)
    while True:
        capture = k4a_playback.get_next_capture()

        if capture is not None:
            color_image = capture.color
            if color_image is not None:
                color_image_data = color_image.data
                color_image_data = np.array(color_image_data, copy=False)
                color_image_bgr = cv2.cvtColor(color_image_data, cv2.COLOR_BGRA2BGR)
                cv2.imshow('Azure Kinect RGB', color_image_bgr)
`
That's my code。
The problem arises when, on the first frame, the color result in the capture object is None
[image]
On the second frame, the color results in the capture object are shown below
[image]
The final realistic image is shown below
[image]
Why is that?
Ask for Help

lpasselin, 2023-07-21:
Is it possible that the depth sync is off? So first frame has depth but no color?

shagren, 2023-07-24:
It's normal behavior. Very often first frame(s) missed color data.
I think it can happens in middle of mkv too.

Cheeee00, 2025-03-04:
通过在packback.py文件中
def get_next_capture(self):    位置，添加：print(self.configuration["color_format"])
输出 ImageFormat.COLOR_MJPG
在原文件中，添加对应的解码：color_image = cv2.imdecode(color_image, cv2.IMREAD_COLOR)