etiennedub/pyk4a #218: cap.color and cap.transformed_depth are not sync

visonpon, 2023-10-27:
Hi @etiennedub , after get rgb and depth, i tried to save them, but there exist sync problem between them, the depth seems always slow than rgb. Hope you can give can advice, thanks~

lpasselin, 2023-10-27:
Can you share the images and the script used to configure the device? They are supposed to be sync, unless you didn't configure it properly. Also since they are not using the same cameras, there is a slight difference.

etiennedub, 2023-10-27:
@visonpon Is there a fixed time offset between the RGB and depth frames? Or the time difference increase with the video duration?
The former would be expected, but not the latter

visonpon, 2023-10-28:
I set config like below
[image]

after overlap the dilated depth to rgb, and you can see that in first frame the depth can overlap suitable， but in second frame the depth not in the proper position and lead to unnatural seg

[image]