etiennedub/pyk4a #225: how to set the iso like whitebalance or exposure in pyk4a.py?

visonpon, 2024-01-23:
Hi, thanks for sharing this wonderful project, I have tried to modify the code to set to iso easily like whitebalance/exposure, but it didn'r work. Hope you can give some help, thanks~

etiennedub, 2024-01-23:
Hi,

It's not supported you get only get the ISO value from a [capture](https://github.com/etiennedub/pyk4a/blob/master/pyk4a/capture.py#L78).

Also, the ISO is set differently from the other parameters. It's not a camera setting but an [image/capture setting on color image](https://github.com/etiennedub/pyk4a/blob/master/pyk4a/capture.py#L78).

I cannot implemented it myself, but if you want you to add it yourself, you can add a call in `PyK4ACapture` that expose [k4a_image_set_iso_speed](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga5a64f64515ff4dff317fec92a8427f45.html#ga5a64f64515ff4dff317fec92a8427f45).

visonpon, 2024-01-24:
ok, i will try again and see if it works, thank you~

closed 2024-01-24 by visonpon