etiennedub/pyk4a #227: Synchronize Multiple Camera

zhoujun-7, 2024-12-29:
Hi, Etienne! 

Thank you for this helpful python wrapper. The code is clear with single camera. But there is not example for multiple camera. Could you please kindly give a example to synchronize multiple cameras using "wired_sync_mode"?

Thanks
Jun

shagren, 2024-12-30:
https://github.com/etiennedub/pyk4a/issues/203#issuecomment-1449635371

zhoujun-7, 2024-12-30:
Hi Shagren,

Thank you for your replying. I had tried your example script. The script includes multiple cameras and works well. 
However, with the limited number of USB ports, sometimes it's impossible to put all cameras into a single script. When I separate each camera into a individual script. The time synchronization is failed.

Here are the codes.
Master:
`
config1 = Config(..., wired_sync_mode=WiredSyncMode.MASTER)
device1=PyK4A(config=config1, device_id=device_id)
device1.start()
while True:
   capture1 = device1.get_capture()
`
Subordinate:
`
config2 = Config(..., wired_sync_mode=WiredSyncMode.SUBORDINATE, subordinate_delay_off_master_usec=200)
device2=PyK4A(config=config2, device_id=device_id)
device2.start()
while True:
   capture2 = device2.get_capture()
`

The Subordinate script is launched firstly, then the Master script launched subsequently.

Looking forward your reply!

Thanks,
Jun

shagren, 2025-01-09:
@zhoujun-7 , you should counts `start_timestamp_offset_usec`, see `PyK4APlayback.configuration`.
Each camera will start with some delay from "start point". This offset contains this delay.